mXlab/morphoses #50: Les données cessent parfois d'arriver des ESP

sofian, 2022-01-18:
Dans le cas du proto 3A et 3B la même situation se produit de temps à autre. Suite à une interruption du programme morphoses.py, on ne reçois plus les données des IMU. Ceci peut se produire tant du côté du IMU sur le côté que du IMU sur le "main". Peut-être relié au mode "sleep"?

sofian, 2022-08-10:
Je confirme que j'ai toujours ce problème qui arrivent de manière sporadique. Le board ne répond plus, même en envoyant /reboot ou /stream rien ne se passe.

sofian, 2023-01-11:
Le problème est survenu à nouveau de manière fréquente en présence de trois robots. Il faut absolument le régler.

Idée:s 
- utiliser les couleurs de lumière pour voir si le robot "gèle" ou si c'est les données seulement
- essayer d'utiliser MQTT à la place

gapzap, 2023-01-11:
J'essaierais de sniffer le WIFI avec Wireshark pour voir qu'est-ce qui bloque! https://www.wireshark.org/download.html
installe WS sur l'ordi qui reçois le data de tout les ESP... aussi peut-être qu'il y plus d'interférence a l'UQAM côté WIFI???

sofian, 2023-01-11:
Ok j'ai installé wireshark mais il va falloir que tu m'expliques comment le configurer!

sofian, 2023-01-11:
J'ai essayé de tracer l'erreur. Sur un des boards au moment de lire les IMU il y a un problème étrange et ça bloque, puis ça reboote; sur le sériel ça affiche le message d'erreur:

```

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x4008f9ef  PS      : 0x00060036  A0      : 0x8008e030  A1      : 0x3ffb08c0  
A2      : 0x3ffbf618  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x3ffbf608  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x80000000  
A14     : 0x007bf618  A15     : 0x003fffff  SAR     : 0x00000020  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40089c30  LEND    : 0x40089c46  LCOUNT  : 0xffffffff  

Backtrace: 0x4008f9ec:0x3ffb08c0 0x4008e02d:0x3ffb0900 0x4008c628:0x3ffb0930 0x4008c5d8:0xa5a5a5a5 |<-CORRUPTED
```

sofian, 2023-01-11:
Je confirme le même problème sur le robot 3.

sofian, 2023-01-11:
Ok fausse alerte, j'ai réussi à corriger ce problème de "core dump" c'était dû à une fonction ```bool processIMU()``` qui ne retournait pas de bool... le compilateur ne me disait rien mais au runtime ça faisait un crash.

sofian, 2023-01-11:
Ok j'ai commencé à tracer les messages avec wireshark, ce qui confirme le phénomène observé avec ```oscdump``` ie. à un moment donné le ESP cesse complètement d'envoyer des messages.

J'ai constaté qu'en appelant la fonction ```setPixels(...)``` cela fait également crasher et cause alors un reboot:

```
rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

Même problème donc si j'envoie le message OSC ```/rgb``` qui appelle cette fonction... Mais c'est un autre problème (car il s'agit d'un "reboot" et non pas d'un "blocage") ie. ça n'explique pas l'interruption des données.

sofian, 2023-01-11:
Le problème était lié à mon algo de reconnection à MQTT qui bouclait indéfiniment. Par contre, je remarque que le signal "lâche" de temps en temps, donc il y a aussi du travail à faire pour améliorer les conditions de transmission du wifi.

commit f105f8f3 (2023-01-11): Prevented mqtt reconnection from hanging out (closes #50).
closed 2023-04-17 by sofian