mXlab/morphoses #52: Il arrive souvent que le balancier fasse un "flip" dans le proto 3b

sofian, 2022-01-20:
Ceci arrive en particulier avec la "skin" 2 qui est plus bulbeuse et plus lourde. Il faudrait probablement simplement ajouter du poids sur le balancier.

gapzap, 2022-01-20:
essaie de ralentir la vitesse d'avancement (-10 a 15% disons) question que le momentum du contrepoids est le temps de faire effet.

sofian, 2022-02-24:
@gapzap Je pense que nous opérons déjà à une vitesse réduite mais je pourrai faire des tests supplémentaires.

sofian, 2022-02-24:
@gapzap Nous avions pris les notes suivantes: "3B tendance à fliper = surement du au fait que le poid était débalancé à cause du skin 2 + pas mis le « bouchon » sur l’autre pôle. Au besoin on pourait quand même en ajouter plus sur les côtés de la batterie"

sofian, 2022-03-29:
Peut-être relié au fait qu'il n'y avait pas de bouchon au moment où on a observé ceci. A monitorer.

sofian, 2022-05-30:
Je ferme ce bug pour le moment, voir si cela se reproduit avec le nouveau design.

closed 2022-05-30 by sofian