mXlab/morphoses #77: Eteindre la RTLS tag du robot en situation de low/critical energy

sofian, 2022-08-09:
Pour suivre sur le #45 : il demeure la RTLS tag à éteinder. En fait on pourrait en situation de voltage critique couper tout le système de manière "hardware" (eg. avec un MOSFET). Pour le moment je mets les composantes en mode "sleep" mais la tag continue de consommer.

gapzap, 2022-08-18:
Je réalise aujourd'hui qu'il faut un vrai power cut-off, comme sur les LIPO ou la batterie elle même coupera le jus!
controler avec un mosfet est problématique car en coupant le jus sur tout(sur le 12v) le ESP32 ne pourra plus se réveiller....
Plutot j'opterais pour une solution comme les LIPO, indépendant et avec hystérésis, j'ai justement la solution, circuit très simple que j'ai fait pour qqu'un d'autre...avec le chip TL431,  je devrai l'ajuster pour le 12v (design avait été pour du 20v).

Le circuit fonctionne comme suit: la batterie chargée présente un voltage max et active le mosfet de sortie. dès que le voltage baisse et passe le niveau défini Vmin (minimum voltage) pour une batterie NIMH , la sortie est couper, plus de jus tant que le voltage ne monte au dessus d'un autre niveau défini Vrecharge, qui est supérieur a Vmin.... 

je regarde ça pour intégration...

sofian, 2022-09-04:
Oui c'est une excellente idée!