mXlab/morphoses #114: A la fin de la journée les robots semblent éprouver plus de difficultés

sofian, 2024-03-17:
Par exemple, le robot 3 commence à ne plus répondre aux commandes de moteurs ou tourne en boucle.
Je soupçonne soit:
* un problème relié à la baisse de voltage dans les batteries
* un problème relié à la fatigue ou surchauffe des moteurs

Solutions à explorer:
* logger la température des moteurs
* éteindre les robots à la fin de la journée et leur donner un "break" puis voir si ça fonctionne ensuite

sofian, 2024-04-10:
https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/

Extrait:
"The DYNAMIXEL can protect itself by detecting dangerous situations that could occur during the operation. Each Bit is inclusively processed with the ‘OR’ logic, therefore, multiple options can be generated. For instance, when ‘0x05’ (binary : 00000101) is defined in Shutdown(63), DYNAMIXEL can detect both Input Voltage Error(binary : 00000001) and Overheating Error(binary : 00000100). If those errors are detected, [Torque Enable(64)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#torque-enable64) is cleared to ‘0’ and the motor’s output becomes 0 [%].

REBOOT is the only method to reset [Torque Enable(64)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#torque-enable64) to ‘1’(Torque ON) after the shutdown.

Check [Alert Bit(0x80)](https://emanual.robotis.com/docs/en/dxl/protocol2/#error) in an error field of Status Packet or a present status via [Hardware Error Status(70)](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#hardware-error-status70). The followings are detectable situations."

sofian, 2024-04-10:
J'ai ajouté du code (voir motors::checkTemperature()) pour gérer le overheating avant qu'on atteigne la limite de 80 C qui cause alors un "overheating error" et disable la torque de manière brutale. Ce n'est pas encore testé par contre. Il resterait à loguer la température des moteurs pour voir ce qu'il en est.

sofian, 2024-05-16:
@etiennemontenegro J'ai eu un problème similaire à la fin de la journée avec le behavior "go_to_y". Le robot 1 ne bougeait plus et ne semblait plus répondre. Il est possible que quand les batteries commencent à être faibles, ça commence à poser des problèmes. Il serait peut-être important que les robots soient rechargés sur l'heure du midi.

etiennemontenegro, 2024-05-16:
Peut-être qu'il faudrait essayer de mettre les robots en light sleep lorsqu'ils sont en mode idle pour économiser un peu de batterie. Je ne sais pas si  ça ferait un changement ou ce serait une économie négligeable

sofian, 2024-05-16:
En effet, ça vaut le coup d'essayer, je pense qu'on n'a pas grand chose à perdre.

sofian, 2024-07-12:
Plusieurs changements ont été faits pour tenter de corriger ce problème et je pense qu'on peut le fermer.

closed 2024-07-12 by sofian